martin-ger/lwip_nat_arduino #1: Error in compiling

h-g93, 2019-03-18:
**hey,
I tried to compile the WiFiNatRouter sketch and I get an error:**

_Build options changed, rebuilding all
/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 1: syntax error: unexpected end of file
make: *** [build/api/api_lib.o] Error 2
exit status 2
Error compiling for board NodeMCU 1.0 (ESP-12E Module)._

help, please

martin-ger, 2019-03-18:
If you Switch Back to the original lwip, does it work?

h-g93, 2019-03-18:
No,
when I try in another computer i get:
_WiFiNatRouter:3:28: error: lwip/lwip_napt.h: No such file or directory
 #include "lwip/lwip_napt.h"
                            ^
compilation terminated.
exit status 1
lwip/lwip_napt.h: No such file or directory_

martin-ger, 2019-03-19:
Okay, the second error is "normal", as the lwip_napt.h is not part of the standard liblwip. Please check the following:
- use the second (standard) setup, chose a builtin example, let's say the Basic Webserver, and compile with the "lwip 1.4 from sources" option.
- if this fails, you have some problems with your Arduino installation
- if this works, try to replace the lwip with my extension - should still work
- if yes, you are ready to go on with NAT...
- if

h-g93, 2019-03-19:
when I try to upload example "HelloServer" with lwip variant "V2 Lower Memory (no features)" its work
when I try to upload example "HelloServer" with lwip variant "lwip 1.4 from sources" it doesn't  work and I get this error:
_exec: "make": executable file not found in %PATH%
Error compiling for board NodeMCU 1.0 (ESP-12E Module)._

Must to say that I didn't find the exact directory of ".../packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/"
but my directory is:
...\Documents\Arduino\hardware\esp8266com\esp8266\tools\sdk

martin-ger, 2019-03-19:
Path is okay. Problem is, that you are compiling on Windows and your environment doesn't have the make tool. That's not a specific issue of this lib, but a general issue with the Arduino on Windows. Could you try on Linux? Maybe Google helps...?

h-g93, 2019-03-20:
hey,
I Succeeded to recompile (in Windows) the lib and i get a new "liblwip_src.a" file and I put him in the lib folder.
it still not working, the same error. Do I have something more to do?

martin-ger, 2019-03-20:
You will need the includes from my lib. Replace the lwip/include drectory with the one from my repo.

h-g93, 2019-03-20:
what do you mean i need the includes from your lib?
I replace the lwip folder like you write, run the MakeFile  and replace the "liblwip_src.a".
 in addition I run "fix_sdk_libs.sh" like write in the README file (in lib folder)

_## Updating SDK libraries

- Copy .a files from SDK `lib` directory to this directory
- Run `fix_sdk_libs.sh`_

I did this ugly to step up the direction error
code snipped :
_#include <ESP8266WiFi.h>

#include "C:/Users/hezi/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/lwip_napt.h"

#include "C:/Users/hezi/Documents/Arduino/hardware/esp8266com/esp8266/tools/sdk/lwip/include/lwip/app/dhcpserver.h"_

and I get:
_.....
exit status 1
'IP_NAPT_MAX' was not declared in this scope
........_

h-g93, 2019-03-21:
I success to solved this, i think this is a bug because need to add 
#include <ESP8266WiFi.h>
#define IP_NAPT 1
#define IP_FORWARD 1 

now I have this problem a lot of time:
c:/users/hezi/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\hezi\AppData\Local\Temp\arduino_build_370994\sketch\WiFiNatRouter.ino.cpp.o:(.text.setup+0x44): undefined reference to `ip_napt_init'

martin-ger, 2019-03-21:
Values are in lwip_napt.h:

```
/* Default size of the tables used for NAPT */
#define IP_NAPT_MAX 512
#define IP_PORTMAP_MAX 32
```

You are obviously not linking against the correct lib or you didn't include lwip_napt.h correctly.

h-g93, 2019-03-21:
You are right but have this:
#if IP_FORWARD
#if IP_NAPT

/* Default size of the tables used for NAPT */
#define IP_NAPT_MAX 512
#define IP_PORTMAP_MAX 32
.
.
#endif /* IP_NAPT */
#endif /* IP_FORWARD */

that make to not compile the header.

so i dont know what to do i have a lot of problems
Do you think that if i do this project on Linux it will be easier?
what do you recommend?

ek20002, 2019-04-18:
hi . i have same error in compiling 
exec: "make": executable file not found in $PATH
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
i am in linux environment
i placed your lwip folder in /home/emad/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/ like you said.
please help me.i really need this for my project

martin-ger, 2019-04-18:
Guess you tried this on Windows?

ek20002, 2019-04-18:
> Guess you tried this on Windows?

no .i am in linux.kubuntu 18.04
i do evrything said in repasetory but still have this error

martin-ger, 2019-04-19:
If you type "make" on the commandline - does ist find the command? If not you probably have to install it.

ek20002, 2019-04-19:
> If you type "make" on the commandline - does ist find the command? If not you probably have to install it.

thats worked. i install make and i can compile and upload it on my nodeMCU .thank you.

Nocold, 2019-04-22:
Hi . i have same error in compiling, on Windows

I install cygwin/make.exe and set path into PATH

then, the arduino ide run:
```
make -C "C:\\Users\\Nocold\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.5.0/tools/sdk/lwip/src" install "TOOLS_PATH=\"C:\\Users\\Nocold\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-\""
```
and returns:
```
/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 1: syntax error: unexpected end of file
```

I execute the command on cygwin-terminal, it terminated at:
```
"C:\Users\Nocold\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-"gcc -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -Ibuild -I/cygdrive/c/Users/Nocold/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/include -I/cygdrive/c/Users/Nocold/Documents/ArduinoData/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/lwip/include  api/sockets.c -o build/api/sockets.o
api/sockets.c:41:22: fatal error: lwip/opt.h: No such file or directory
 #include "lwip/opt.h"
                      ^
compilation terminated.
```

help, please.

ek20002, 2019-04-25:
hi again .i have quastion. how much is muximum speed of this?
i cant get more than 50 KB!!!!!!

martin-ger, 2019-04-25:
Depends on the signal strength ond loss rate - under good conditions about 4-5 MBit are possible.

ek20002, 2019-04-29:
hi. sorry its me again.i have another quastion.
struct pbuf {
  /** next pbuf in singly linked pbuf chain */
  struct pbuf *next;

  /** pointer to the actual data in the buffer */
  void *payload;

  /**
   * total length of this buffer and all next buffers in chain
   * belonging to the same packet.
   *
   * For non-queue packet chains this is the invariant:
   * p->tot_len == p->len + (p->next? p->next->tot_len: 0)
   */
  u16_t tot_len;

  /** length of this buffer */
  u16_t len;

  /** pbuf_type as u8_t instead of enum to save space */
  u8_t /*pbuf_type*/ type;

  /** misc flags */
  u8_t flags;

  /**
   * the reference count always equals the number of pointers
   * that refer to this pbuf. This can be pointers from an application,
   * the stack itself, or pbuf->next pointers from a chain.
   */
  u16_t ref;
  
  /* add a pointer for esf_buf */
  void * eb; 
};

its pbuf struct in pbuf.h file .iwant to know what is unit of len and tot_len attributes of this struct?and sum of  len   can be used to meter trafic that pass from esp?

martin-ger, 2019-04-29:
uint is bytes, len is the size of this buffer and tot_len the payload length of the complete buffer chain (in almost any case you have yu´´just one buffer here). You can add up tot_len to meter the traffic.

guigui9, 2019-05-29:
Hello,
here is the procedure to install and compile the library in Windows (7, 8.1 and 10).
(thanks for the file "liblwip_src.a").

```
1) Downloading files :
Download Arduino IDE 1.8.9 (portable) :
  https://downloads.arduino.cc/arduino-1.8.9-windows.zip
Download "lwip_nat_arduino-master.zip" :
  https://codeload.github.com/martin-ger/lwip_nat_arduino/zip/master
Download "hh2.golden.exe" (402 bytes) :
  https://github.com/pts/pts-tinype/raw/master/hh2.golden.exe

2) Configure Arduino with ESP8266 library :
Unzip "arduino-1.8.9-windows.zip" to "C:\App\arduino-ide\"
Makedir "portable" in "C:\App\arduino-ide\"
Create new file : "C:\App\arduino-ide\portable\preferences.txt"
Add line with :
  boardsmanager.additional.urls=http://arduino.esp8266.com/stable/package_esp8266com_index.json
Open IDE Arduino and install ESP8266 library in menu :
  Tools / Boards: "Arduino..." / Boards Manager

3) Install the lwip_nat_arduino library :
Copy and rename "hh2.golden.exe" to "C:\App\arduino-ide\make.exe"
  (to have an empty "make" utility)
Open "lwip_nat_arduino-master.zip"
Next operation in folder :
  "C:\App\arduino-ide\portable\packages\esp8266\hardware\esp8266\2.5.2\tools\"
Rename lwip\ with lwip.orig\ in sdk\
Unzip  lwip\                 in sdk\lwip\
Unzip  liblwip_src.a         in sdk\lib\

4) Change the compilation mode :
Menu : Tools / lwIP Variant: "v1.4 Compile from source"
Menu : Tools / Erase Flash: "Sketch + WiFi Settings"
```