martin-ger/lwip_nat_arduino #2: newbie for real napt_init

joe318, 2019-03-21:
Hello,
i'm trying to find a solution with the functionality of repeater, i found your library, i try but he tell me:
IP_NAPT_MAX and IP_PORTMAP_MAX not declared, my problem is what i have to declare?

Regards,

joe318, 2019-03-21:
Ok i find why but now i have this
exec: "make": executable file not found in %PATH%

martin-ger, 2019-03-21:
Values are in lwip_napt.h:

```
/* Default size of the tables used for NAPT */
#define IP_NAPT_MAX 512
#define IP_PORTMAP_MAX 32
```

You are compiling on Windows?

joe318, 2019-03-21:
yesss lol f... microsoft i know, idk why this make path want

martin-ger, 2019-03-21:
In https://github.com/esp8266/Arduino/issues/3143 Ivan Grokhotkov (igrr) says:
"i'm afraid windows users need to install mingw and use make to compile LwIP. "

joe318, 2019-03-21:
omg, maybe i try or i turn on one raspberry, what you think?

martin-ger, 2019-03-21:
I do all this on a Linux in a VM...

joe318, 2019-03-21:
oh yes i forget this idea

joe318, 2019-03-21:
omg im trying hard but he bother me with windows, just a question, if i make with linux need, i need continue use linux for program the arduino right?

i'm in one crazy idea in my head, one mix between websocket external server and esp8266, want manage my home with and i need one way for communicate with few esp so musch far of my router, thats why im trying use ur lwip, i try painlessmesh too, can work but complicate for manage which esp need do something and answer the states of my relays and more.
For the socket i use socket.io, with little mix of sql, javascript and php, everything work well, now need one mesh but need ota too.
maybe u have another idea for i can do what i want

Regards,

joe318, 2019-03-21:
The last answer of my make try
make: *** [build/api/api_lib.o] Erreur 1

he do me that with others but i solve it, just this file idk where he is, cant find.

martin-ger, 2019-03-22:
it is in lwip/src/api

joe318, 2019-03-22:
yes and no, he ask api_lib.o not api_lib.c

joe318, 2019-03-22:
omgggg i put virtualbox, install ububtu 18, installing arduino ide, esp8266 library, without problem, but i have one sooooo strange problem, my brain is burningggg.
i don't know where is the folder lwip shit, i try with locate, in user and root, i try find with manyyy words like esp8266, lwip, .... nothinggg but i see vcan select the esp8266 inside arduino ide, so the library is here but idk whereeeee it's so crazyyyy lol
which linux you use?

martin-ger, 2019-03-22:
Probably you have a hidden directory ".arduino15" in your home directory.

I use Ubuntu too...

joe318, 2019-03-22:
omg sorry, feel me stupid, so simple f...
Thanks a lot, i'm going test now, i told you if it working

joe318, 2019-03-23:
ok, done, working, with ubuntu in vm
This message just for told you 2 important things !!!
1st sorry for bother you
2nd thanks a lottttt, we need people like youuuuu, kiss babe lol