martin-ger/lwip_nat_arduino #3: How to setup in PlatformIO IDE

Microinstance, 2019-05-02:
Hi, please guide how did I set up this in PlatformIO

martin-ger, 2019-05-02:
Sorry, havn't installed that. Guess you can replace the liblwip.a (just added the binary to the repo) somewhere and also the lwip/include directory. But I havn't tested that...

HelloDB, 2019-05-12:
@Microinstance @martin-ger 
hi！
It's easy on platformio, just put it in the lib folder.
Thanks for a great library.
[image]
[image]

irwanayeah, 2019-05-19:
great, I have success following step by step. thx guys

Syavne, 2019-10-03:
In case someone uses only the terminal and not the IDE (personal case), the library to replace is located in  ~/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip so you have to do something like the following:
`$ mv ~/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip /.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip.bkp`
`$ ln -s lwip_nat_arduino/lwip ~/.platformio/packages/framework-arduinoespressif8266/tools/sdk/lwip`

I figured it out a while after reading this page, thank you guys!.

Seyyed1404, 2024-03-06:
Can I use this example for esp32 in arduino?