martin-ger/lwip_nat_arduino #7: Error compiling for board Generic ESP8266 Module.

RoyvanEmpel, 2019-06-05:
Hello,

I am trying to install the CaptiveHotspot on the ESP8266 but if i try to verify it i get the error: 

exec: "make": executable file not found in %PATH%
Error compiling for board Generic ESP8266 Module.

In the readme it says "Make sure that the 'make' command is installed in your environment" but i don't know how to do this. I have googled it but can't really find anything about it.

I hope somebody can tell me how to fix this.

martin-ger, 2019-06-05:
You probably do this on Windows, right?

You will need a make for Windows, e.g.: https://stackoverflow.com/questions/32127524/how-to-install-and-use-make-in-windows/53228840

RoyvanEmpel, 2019-06-05:
Yes i am tying to do this on a windows machine. and i looked at the stackoverflow that you send but when i try to download the GNUWin32 and got the make.exe from the /GNUWin32/bin/ and i tried to put it in the arduino directory but that doesn't work and now it is giving me this error: exit status -1073741515 but i can't really find anything about this :/.

PS: I am a real arduino noob and i am doing this for a school project.

martin-ger, 2019-06-05:
Sorry, not an expert in that: I do all my compilations on Linux. I know, it can be done on Windows and people did it, but I have no experience in installing a make environment there.

guigui9, 2019-06-05:
Hello,
my method works to compile on Windows :
[https://github.com/martin-ger/lwip_nat_arduino/issues/1](https://github.com/martin-ger/lwip_nat_arduino/issues/1#issuecomment-496891966)

RoyvanEmpel, 2019-06-05:
Hello,
Thank you @guigui9 it did work for me and i got it compiled and uploaded to my esp8266.

Now i am having a problem with getting the code to work.
I changed the sta_ssid to the name of my network and the sta_password to my wifi password. But when I upload it and look in my router it isn't connected. I also tried to start a hotspot, but also then it isn't connecting or starting its own AP.

Do you maybe know what could be this?

This is the upload log. (Or whatever its called)

> Sketch uses 287396 bytes (32%) of program storage space. Maximum is 892912 bytes.
Global variables use 28788 bytes (35%) of dynamic memory, leaving 53132 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.6
Serial port COM6
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: a0:20:a6:1b:20:17
Uploading stub...
Running stub...
Stub running...
Erasing region (may be slow depending on size)...
Erase completed successfully in 0.2 seconds.
Hard resetting via RTS pin...
esptool.py v2.6
Serial port COM6
Connecting....
Chip is ESP8266EX
Features: WiFi
MAC: a0:20:a6:1b:20:17
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Flash params set to 0x0040
Compressed 291552 bytes to 209066...

> Writing at 0x00000000... (7 %)
Writing at 0x00004000... (15 %)
Writing at 0x00008000... (23 %)
Writing at 0x0000c000... (30 %)
Writing at 0x00010000... (38 %)
Writing at 0x00014000... (46 %)
Writing at 0x00018000... (53 %)
Writing at 0x0001c000... (61 %)
Writing at 0x00020000... (69 %)
Writing at 0x00024000... (76 %)
Writing at 0x00028000... (84 %)
Writing at 0x0002c000... (92 %)
Writing at 0x00030000... (100 %)
Wrote 291552 bytes (209066 compressed) at 0x00000000 in 18.4 seconds (effective 126.8 kbit/s)...
Hash of data verified.

> Leaving...

> Hard resetting via RTS pin...

martin-ger, 2019-06-05:
Do you see anything on the serial output? Should be some info here...

RoyvanEmpel, 2019-06-05:
It is doing this and keeps making those dots. :/

[image]

martin-ger, 2019-06-05:
Mhhh... for some reason it doesn't get a connection to the uplink AP.
Are you sure my_ssid and my_password are correct?
It works for other Arduino samples?

RoyvanEmpel, 2019-06-05:
the esp fully works for other project like just creating a AP and running a local webserver on that ap. I am going to just try it on another hotspot who knows what will happen.

RoyvanEmpel, 2019-06-05:
I am going to try to upload some old code i know is working.

RoyvanEmpel, 2019-06-05:
So i just uploaded some older code which creates an access point without any problems :/

martin-ger, 2019-06-05:
Creating the AP is not the problem. The program hangs when it tries to connect to the uplink AP as STA in line 217-220. AP creation is done afterwards...

guigui9, 2019-06-05:
Hello,
I had connection failures. But the Wifi password must not be less than 8 characters.
Another option allowed me to correct some program errors,
menu : Tools / Erase Flash: "All Flash Contents"

RoyvanEmpel, 2019-06-05:
Hmm I see.

Changing the Erase Flash to "All Flash Contents" Doesn't change anything and i am still getting the same problem.

So this is the full code currently (at least in the captivehotspot.ino)
I don't know if i accedentally broke something.

[https://pastebin.com/raw/tiG8qqUF](url)

martin-ger, 2019-06-05:
With the correct credentials the program runs absolutely fine for me.

Is your uplink AP really "SlechtWifi"/"wachtwoord"? This is not the setting for the ESP-AP...

RoyvanEmpel, 2019-06-05:
so with uplink AP you mean my hotspot/wifi right?

RoyvanEmpel, 2019-06-05:
[image]

martin-ger, 2019-06-05:
Yes, your wifi. 
Other programs can connect?

RoyvanEmpel, 2019-06-05:
I can connect with my phone yes but its also not working on my home network or my mobile phone hotspot.

Do i maybe have a setting wrong? 

[image]

martin-ger, 2019-06-05:
I don't see any issue there. 

When you try to compile the "ESP8266WiFi/WiFiClient" example - does this work?

RoyvanEmpel, 2019-06-05:
Well I am getting the same result :/
So its not a problem with the code :/

[image]

guigui9, 2019-06-05:
Your Internet box may be using WPS mode with mac address filtering.

martin-ger, 2019-06-05:
Some issue with you WiFi? Can you try another one. E.g. your Smartphone as hotspot?

RoyvanEmpel, 2019-06-05:
So i have tried it on my mobile hotspot from my phone, my laptop hotspot and my normal wifi router i really have no clue what it could be but none of them are working.

Tomorrow i will try the script on the ESP of a friend of me and if that doesn't work i really don't know :/

RoyvanEmpel, 2019-06-06:
@martin-ger After trying on the ESP of my friend and tried it on the hotspot of my friend but i can't get it to work it just keeps printing the dots

martin-ger, 2019-06-06:
Even when you use the standard example? No idea...

guigui9, 2019-06-06:
It may not change anything, but my program works fine with the settings :
Flash Mode: "DOUT (compatible)"
Crystal Frequency: "26 MHz"

RoyvanEmpel, 2019-06-09:
Sorry for not reacting for a few day's. I have tried the settings with the flash mode and the crystal frequency and it just started working :) thanks for helping

martin-ger, 2019-06-09:
Good to read! :-)

closed 2019-06-09 by martin-ger

JayantBenjamin, 2019-10-08:
> Hello,
> Thank you @guigui9 it did work for me and i got it compiled and uploaded to my esp8266.
>
> Now i am having a problem with getting the code to work.
> I changed the sta_ssid to the name of my [...]

Hi! How did you solve the "make" problem?
Error says exec: "make"; executable file not found in %PATH%
Another error I'm getting is "Multiple Libraries Found "

vienbu, 2021-05-19:
Hello every guy, i'm newbie and i make this error, i don't resolve, can any one help me, please? 
Arduino: 1.8.15 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\Firebase.cpp: In member function 'int FirebaseRequest::sendRequest(const string&, const string&, char*, const string&, const string&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\Firebase.cpp:78:1: warning: no return statement in function returning non-void [-Wreturn-type]

   78 | }

      | ^

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\Firebase.cpp: In member function 'void FirebaseStream::startStreaming(const string&, const string&, const string&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\Firebase.cpp:91:16: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   91 |   analyzeError("STREAM", status, path_with_auth);

      |                ^~~~~~~~

In file included from C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseCloudMessaging.cpp:1:

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseCloudMessaging.h:82:3: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

   82 |   const void AddToJson(const FirebaseCloudMessage& message, JsonObject& json) const;

      |   ^~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp: In member function 'String FirebaseArduino::push(const String&, const ArduinoJson::JsonVariant&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:65:54: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   65 |   int status = req_.get()->sendRequest(host_, auth_, "POST", path.c_str(), buf);

      |                                                      ^~~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:65:7: warning: unused variable 'status' [-Wunused-variable]

   65 |   int status = req_.get()->sendRequest(host_, auth_, "POST", path.c_str(), buf);

      |       ^~~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp: In member function 'void FirebaseArduino::set(const String&, const ArduinoJson::JsonVariant&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:94:41: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

   94 |   req_.get()->sendRequest(host_, auth_, "PUT", path.c_str(), buf);

      |                                         ^~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp: In member function 'void FirebaseArduino::getRequest(const String&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:101:41: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

  101 |   req_.get()->sendRequest(host_, auth_, "GET", path.c_str());

      |                                         ^~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp: In member function 'virtual void FirebaseArduino::remove(const String&)':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:147:41: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

  147 |   req_.get()->sendRequest(host_, auth_, "DELETE", path.c_str());

      |                                         ^~~~~~~~

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp: In member function 'virtual const String& FirebaseArduino::error()':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseArduino.cpp:201:32: warning: returning reference to temporary [-Wreturn-local-addr]

  201 |   return error_.message().c_str();

[... 94 lines cut ...]
C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.cpp: In member function 'ArduinoJson::JsonVariant FirebaseObject::getJsonVariant(const String&) const':

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.cpp:91:26: warning: 'ArduinoJson::JsonObject& ArduinoJson::Internals::JsonVariantCasts<TImpl>::asObject() const [with TImpl = ArduinoJson::JsonVariant]' is deprecated: use as<JsonObject>() instead [-Wdeprecated-declarations]

   91 |     json = json.asObject().get<JsonVariant>(start);

      |                          ^

In file included from C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.h:24,

                 from C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.cpp:17:

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src/ArduinoJson.h:246:28: note: declared here

  246 |   FORCE_INLINE JsonObject &asObject() const {

      |                            ^~~~~~~~

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

martin-ger, 2021-05-19:
You don’t need this lib any more:

 

since release 2.6.0 NAPT/NAT is part of the standard distribution. See also this example:  <https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino> https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino ***

 

 
 

 
 

 
 

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.cpp:71:57: warning: 'const char* ArduinoJson::Internals::JsonVariantCasts::asString() const [with TImpl = ArduinoJson::JsonVariant]' is deprecated: use as<char*>() instead [-Wdeprecated-declarations]

C:\Users\vienp\OneDrive\Documents\Arduino\libraries\firebase-arduino-master\src\FirebaseObject.cpp:91:26: warning: 'ArduinoJson::JsonObject& ArduinoJson::Internals::JsonVariantCasts::asObject() const [with TImpl = ArduinoJson::JsonVariant]' is deprecated: use as() instead [-Wdeprecated-declarations]

91 | json = json.asObject().get(start);

  |                          ^

vienbu, 2021-05-20:
@martin-ger  i want to connect with firebase realtime, if i don't add the lib firebase, i can't send data from sensor to firebase realtime, so what do I have to do? Sr @martin-ger because I'm new, I lack understanding of IDE , plz forgive me for my shortcomings :))