martin-ger/lwip_nat_arduino #8: External captive portal

akshar001, 2019-07-13:
Hello,
This is a great thing to do,
i thank you completely for effort,
Now what if we want to redirect the request to any other internet page?
Like external captive portal running in server,
How would we do that?

martin-ger, 2019-07-13:
Think, it should be enough to change the redirects in l130

akshar001, 2019-07-13:
Can you just let me know how to do that, I mean a way to do it. I will try to do it and as well contribute to your library for other people as well,

martin-ger, 2019-07-13:
Lines 130 - 136 and lines 169 - 173 in  CaptiveHotspot.ino are the redirects of any HTTP request to the local address and the rewrite of the replies. This has to be changed to the IP of the other server.

When that server at this IP has handled the request successfully, it has to tell the ESP to add it to the allowed_macs array. Could be done by a separate service on the ESP or perhaps via a central MQTT broker...?

akshar001, 2019-07-16:
Yes they redirect any kind of http request to requested ip address. I understand your concern about how to add allowed_mac address to local esp if we are redirecting to a server, Yes we can use mqtt or local http get request to server, there are many ways, now what i learnt that this code modify the ip address of the requests to our desire ip, but i want to find out how we are going to modify the pages thing like what if we want to redirect to specific page, where we need to kind a change whole http url itself in request not just ip address.

closed 2019-07-16 by akshar001

martin-ger, 2019-07-16:
Okay, what about the following idea:
- leave the local server and let it redirect any request (HTTP 302) to the external server.
- here you can rewrite the request to even contain the (encoded) MAC (see line 272 ff.)
- of course it should have an additional rule in check_packet_in() to let HTTP traffic to the external server pass.
- with that you can even have a multipage dialog with the server before authentication.
- if you distribute the allowed MAC via MQTT, you can even inform other ESPs about it, so once a station is authenticated, it can use any ESP (of possibly many) as router without further interaction.

akshar001, 2019-07-16:
Yes it would be great,
I tried one thing that already,
But i was using esp8266 with https and ESP got restarts somewhere