martin-ger/lwip_nat_arduino #13: About Arduino platform NAT development compilation failed!

JackMing0, 2019-11-13:
Thank you for your comments!

I transferred the project to the Arduino development platform for operation

Operation:
The following operations are in strict accordance with the blogger: https://github.com/martin-ger/lwip_nat_arduino project installation instructions!
[image]

Question:
The following problems occurred during compilation:

正在编译项目...
make -C "C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip/src" install "TOOLS_PATH=\"C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-\""
使用 1.0  版本的库 ESP8266WiFi 在文件夹： C:\Users\mac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 
exec: "make": executable file not found in %PATH%
为开发板 Generic ESP8266 Module 编译时出错。

The operation process：
        
[image]
[image]
[image]
[image]

martin-ger, 2019-11-13:
On Windows, make ist missing. Maybe this helps: https://github.com/martin-ger/lwip_nat_arduino/issues/1#issuecomment-496891966

JackMing0, 2019-11-13:
The above problem was solved after I added the static library!
It is in compile again appear new problem!

Please help me! Thank you very much!

The questions are as follows:

正在编译项目...
make -C "C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip/src" install "TOOLS_PATH=\"C:\\Users\\mac\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-\""
make: Entering directory '/cygdrive/c/Users/mac/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip/src'
\C:\Users\mac\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-"gcc -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -ffunction-sections -fdata-sections -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -DLWIP_OPEN_SRC -Ibuild -I/cygdrive/c/Users/mac/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/include -I/cygdrive/c/Users/mac/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip/include  api/sockets.c -o build/api/sockets.o
/bin/sh: -c: line 0: unexpected EOF while looking for matching `"'
/bin/sh: -c: line 1: syntax error: unexpected end of file
make: *** [Makefile:26: build/api/sockets.o] Error 1
make: Leaving directory '/cygdrive/c/Users/mac/AppData/Local/Arduino15/packages/esp8266/hardware/esp8266/2.5.2/tools/sdk/lwip/src'
使用 1.0  版本的库 ESP8266WiFi 在文件夹： C:\Users\mac\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 
exit status 2
为开发板 Generic ESP8266 Module 编译时出错。

[image]