martin-ger/lwip_nat_arduino #14: dhcps_set_DNS(WiFi.dnsIP());

parisli0086, 2019-11-26:
Dear all
I meet one problem about the line: dhcps_set_DNS(WiFi.dnsIP());
from feedback of arduino IDE: conversion from 'IPAddress' to 'ip_addr*' is ambiguous
I check everywhere but not find answer.
Thanks for read , prefer your answer.

martin-ger, 2019-11-26:
You might want to try that with the standard esp8266 Arduino. Since release 2.6.0 it support NAPT: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

parisli0086, 2019-11-27:
> You might want to try that with the standard esp8266 Arduino. Since release 2.6.0 it support NAPT: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

Thank you very much for your help. I already try your link, it's very good solve my problem. thank you again and best wishes for you.