martin-ger/lwip_nat_arduino #15: ESP01

PaleoAgTech, 2020-02-20:
Is it possible to use that with ESP01? I try upload it but chip gets restarting. I tryed another powersuply with cap but still same problem.

martin-ger, 2020-02-20:
It works on the ESP01, but perhaps you screwed up the config for the board?

But try the latest ESP8266 Arduino - should be the easiest:

This lib is somewhat obsolete because of the recent update of ESP8266 Arduino https://github.com/esp8266/Arduino. Since release 2.6.0 NAPT/NAT is part of the standard distribution. See also this example: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/RangeExtender-NAPT/RangeExtender-NAPT.ino

PaleoAgTech, 2020-02-20:
I got it too work. It was my config screwup. One more question. How can I change my softAP ip from 192.168.4.1 to e.g 10.24.1.1?

martin-ger, 2020-02-20:
You can find a code snipplet here: https://www.esp8266.com/viewtopic.php?f=8&t=8603

PaleoAgTech, 2020-02-20:
Thank you for your help so far. I was looking the code and it looks all logical and fine but to be honest I have no idea how to use it in a sketch :(

martin-ger, 2020-02-20:
Use this code in the setup() of you sketch to manipulate the IP address of the softAP.

PaleoAgTech, 2020-02-20:
I got it almost work now.
Only problem is that it had error with these lines

dhcp_lease.start_ip = ipaddr_addr(start_ip);
dhcp_lease.end_ip = ipaddr_addr(end_ip);

No match for 'operator=' (operand types are 'ip_addr' and 'u32_t {aka long unsigned int}')

And if I remove these I have no internet access.

martin-ger, 2020-02-20:
Try:
dhcp_lease.start_ip.addr = ipaddr_addr(start_ip);
dhcp_lease.end_ip.addr = ipaddr_addr(end_ip);

PaleoAgTech, 2020-02-20:
It worked!
Only now I have no internet access in that AP network

martin-ger, 2020-02-20:
Did you do the:
```
  // Initialize the NAT feature
  ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);

  // Enable NAT on the AP interface
  ip_napt_enable_no(1, 1);
```
afterwards?

PaleoAgTech, 2020-02-20:
Well I try but now it keeps connecting and disconnecting. What do I need to add there?

martin-ger, 2020-02-20:
Don't know. You might send me your Setup() Code.

PaleoAgTech, 2020-02-20:
```
void setup()
{

  struct softap_config config;
struct dhcps_lease dhcp_lease;
char* start_ip = "192.168.5.100";
char* end_ip = "192.168.5.105";
struct ip_info info;

// char ssid[32]; //= STA_SSID;
// char password[64]; //= STA_PASS;
// wifi_set_opmode(SOFTAP_MODE);
wifi_set_opmode(STATIONAP_MODE);
wifi_set_phy_mode(PHY_MODE_11N); // maximom power
wifi_softap_get_config(&config); // Get config first.
os_memset(config.ssid, 0, 32);
os_memset(config.password, 0, 64);
os_memcpy(config.ssid, "Uplink", 6);
os_memcpy(config.password, "", 0);
config.authmode = AUTH_WPA_WPA2_PSK;
config.ssid_len = 0;// or its actual length
config.max_connection = 8; // how many stations can connect to ESP8266 softAP at most.

wifi_softap_dhcps_stop();

IP4_ADDR(&info.ip, 192, 168, 5, 1);
IP4_ADDR(&info.gw, 192, 168, 1, 1);
IP4_ADDR(&info.netmask, 255, 255, 255, 0);

dhcp_lease.start_ip.addr = ipaddr_addr(start_ip);
dhcp_lease.end_ip.addr = ipaddr_addr(end_ip);

// IP4_ADDR(&dhcp_lease.start_ip, 192, 168, 4, 100);
//IP4_ADDR(&dhcp_lease.end_ip, 192, 168, 4, 105);

wifi_set_ip_info(SOFTAP_IF, &info);
wifi_softap_set_dhcps_lease(&dhcp_lease);
wifi_softap_dhcps_start();

wifi_softap_set_config(&config);// Set ESP8266 softap config .

 
  Serial.begin(115200);
  Serial.println();
 
  WiFi.mode(WIFI_AP_STA);

  Serial.println("Starting NAT demo");
  
  WiFi.begin(sta_ssid, sta_password);
 
  //WiFi.config(ip, gateway, subnet);

  //Wifi connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(sta_ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("dnsIP address: ");
  Serial.println(WiFi.dnsIP());
  Serial.print("gatewayIP address: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("subnetMask address: ");
  Serial.println(WiFi.subnetMask());

  Serial.println("");
  Serial.println("Configuring access point...");
  WiFi.softAP(ap_ssid, ap_password);
 
   IPAddress ip(192,168,1,200);   
   IPAddress gateway(192,168,1,254);   
   IPAddress subnet(255,255,255,0);   
   WiFi.config(ip, gateway, subnet);
 
  IPAddress myIP = WiFi.softAPIP();

  
  
  Serial.print("AP IP address: ");
  Serial.println(myIP);

 // Initialize the NAT feature
  ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);

  // Enable NAT on the AP interface
  ip_napt_enable_no(1, 1);

  // Set the DNS server for clients of the AP to the one we also use for the STA interface
  dhcps_set_DNS(WiFi.dnsIP());
}
```

This is actually your original code where I added IP changing code

martin-ger, 2020-02-21:
Actually
`IP4_ADDR(&info.gw, 192, 168, 1, 1);`
is wrong, should be the ESP: 192.168.5.1.

The code below is a little bit shorter and works for me:
```
void setup()
{
  struct dhcps_lease dhcp_lease;
  char* start_ip = "192.168.5.100";
  char* end_ip = "192.168.5.105";
  struct ip_info info;
  
  Serial.begin(115200);
  Serial.println();

  WiFi.mode(WIFI_AP_STA);

  Serial.println("Starting NAT demo");
  
  WiFi.begin(sta_ssid, sta_password);
  //WiFi.config(ip, gateway, subnet);

  //Wifi connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(sta_ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.print("dnsIP address: ");
  Serial.println(WiFi.dnsIP());
  Serial.print("gatewayIP address: ");
  Serial.println(WiFi.gatewayIP());
  Serial.print("subnetMask address: ");
  Serial.println(WiFi.subnetMask());

  Serial.println("");
  Serial.println("Configuring access point...");
  WiFi.softAP(ap_ssid, ap_password);
  
  wifi_softap_dhcps_stop();
  
  IP4_ADDR(&info.ip, 192, 168, 5, 1);
  IP4_ADDR(&info.gw, 192, 168, 5, 1);
  IP4_ADDR(&info.netmask, 255, 255, 255, 0);
  
  dhcp_lease.start_ip.addr = ipaddr_addr(start_ip);
  dhcp_lease.end_ip.addr = ipaddr_addr(end_ip);
  
  // IP4_ADDR(&dhcp_lease.start_ip, 192, 168, 4, 100);
  //IP4_ADDR(&dhcp_lease.end_ip, 192, 168, 4, 105);
  
  wifi_set_ip_info(SOFTAP_IF, &info);
  wifi_softap_set_dhcps_lease(&dhcp_lease);
  wifi_softap_dhcps_start();

  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

  // Initialize the NAT feature
  ip_napt_init(IP_NAPT_MAX, IP_PORTMAP_MAX);

  // Enable NAT on the AP interface
  ip_napt_enable_no(1, 1);

  
  // Set the DNS server for clients of the AP to the one we also use for the STA interface
  dhcps_set_DNS(WiFi.dnsIP());
}
```

PaleoAgTech, 2020-02-21:
OMG Sir this actually works. Thank you so much for that!

closed 2020-02-21 by martin-ger