martin-ger/lwip_nat_arduino #16: WiFi Repeat from ethernet adapter based on W5500

fdellagnello, 2020-02-21:
Hi martin-ger,
I'm working on a sketch to stream data from/to a W5500 ethernet adapter (interfaced via spi with the ESP8266 in Arduino IDE) to/from WiFi single remote client. Until now I'm able to create the access point, the Ethernet Server, the WiFi Server and share both UDP or TCP-IP packets. The performance is quite good for my application but the preliminary configuration is really limiting because I need to specify the pre-defined IP Address (not a big issue) and pre-defined server port (big issue). So I'm looking at NAT server function for this framework. 
I'm running your WiFi range extender and it works pretty well, so I just wondering if it is possible to replace the WiFi STA connection with a wired Ethernet one based exactly on W5500. Unfortunately I haven't time until now to study in details your library and how your routine works but before waste time to do that, I'm thinking to ask about any feedback.

Let me know,
Thanks in advance,
Ciao,
Fil McLamb

martin-ger, 2020-02-21:
Difficult: the W5500 implements it's own TCP/IP stack and offers a socket-style interface to the device. This is good, as it makes network programming easier on small MCUs like the AVR, but it hides IP from the controller and it doesn't integrate into the ESPs lwip TCP/IP stack. As far as I understood the W5500 cannot send arbitrary IP packets and thus NAT-ed IP traffic. Maybe it can be done with IPRAW sockets, but at a first look even this was not obvious to me.

That's why I did some experients with the enc28j60 ethernet controller, but this isn't solid right now. (https://github.com/martin-ger/esp-open-lwip).

fdellagnello, 2020-02-22:
I understand, 
your considerations are really valuable. I don't need an internet connection but just a local network so at the end the TCP/IP stack of W5500 should work if I'm able to manage within the WiFi frame. Anyway I start to consider the use of ENC28J60. 

Thanks so much,
Really Appreciate,
Ciao,
F.