martin-ger/lwip_nat_arduino #17: How to config ip_route for pinging another esp?

TridentTD, 2020-02-26:
I have 2 esps as the following scenario

| ESP1-STA 192.168.43.10 | <-> |Home Router 192.168.43.1| <-> | 192.168.43.20 STA-ESP2 |

By this lwip-nat_arduino's liblwip_src.a, how to set any function 
for I can ping ESP2 192.168.43.20  by ESP1 ?

I have tried to set `ip_add_route( ip, mask, gw);`  but it's worked.

Thank you.

martin-ger, 2020-02-26:
All hosts are in the same subnet, they should "see" each other without any route. 

```
ping 192.168.43.20
```
fails?

What kind of ping command do you use on the ESP?

TridentTD, 2020-02-29:
Thanks for the answer and sorry for my late reply.

I change scenario to the following and with the code.
The result of pinging to the  Notebook-STA (192.168.43.6) can't ping.
How to solve the issue?

Thank you.

## scenario
| ESP1-STA 192.168.43.254 | <-> |Home Router 192.168.43.1| <-> | 192.168.43.6 Notebook-STA |

## code for ESP1-STA
```cpp
#include <ESP8266WiFi.h>
#include "lwip/lwip_napt.h"
#include "lwip/app/dhcpserver.h"
#include "lwip/ip_route.h"
#include "user_interface.h"
#include "ping.h"

#define  SSID         "-----------------"
#define  PASSWORD     "-----------------"

#define  AP_SSID      "-----------------"
#define  AP_PASSWORD  "-----------------"

#define NAPT 1000
#define NAPT_PORT 10

void esp_ping(IPAddress ip);

void setup() {
  Serial.begin(115200); Serial.println();
  WiFi.begin(SSID, PASSWORD);
  WiFi.softAP(AP_SSID, AP_PASSWORD);

  ip_napt_init(NAPT, NAPT_PORT);
  ip_napt_enable_no(SOFTAP_IF, 1);

  while(!WiFi.isConnected()) { delay(400); Serial.print("."); }
  Serial.println();
  Serial.print("STA IP : "); Serial.println(WiFi.localIP());
  Serial.print("AP  IP : "); Serial.println(WiFi.softAPIP());

  dhcps_set_DNS(WiFi.dnsIP(0));
  
  esp_ping(IPAddress(192,168,43,6));  // ping to IP of the Notebook-STA.
}

void loop() {
  // put your main code here, to run repeatedly:

}

void esp_ping(IPAddress ip){
  static struct ping_option  ping_opt;

  ping_opt.count        = 4;
  ping_opt.coarse_time  = 1;
  ping_opt.ip           = ip.v4();
  
  ping_regist_recv(&ping_opt, [](void* arg, void *pdata){
    struct ping_option* ping_opt = (struct ping_option*) arg;
    struct ping_resp *  ping_res = (struct ping_resp*) pdata;
      
    if(ping_res->ping_err == -1){
      Serial.printf("Reply from %s : offline\n", 
          IPAddress(ping_opt->ip).toString().c_str() );
    }else{
      Serial.printf("Reply from %s : bytes = %d time = %d ms\n", 
          IPAddress(ping_opt->ip).toString().c_str(), 
          ping_res->bytes, ping_res->resp_time);
    }
  });
  
  ping_start(&ping_opt);
}
```
## Result
[image]

## Notebook-STA by ipconfig
[image]

martin-ger, 2020-02-29:
NAPT should not be involved here at all - it only comes into play, when you try to ping from an additional STA in the AP (192.168.4.x) network.

So this should work. Does it work with the original lwip? With an lwip 2 version? Without these calls?
```
ip_napt_init(NAPT, NAPT_PORT);
ip_napt_enable_no(SOFTAP_IF, 1);
```

TridentTD, 2020-03-01:
Thanks again.

When I tried by the following code with` lwip v2 Lower Memory`,
The result can't ping from ESP (192.168.43.254) to the Notebook (192.168.43.6).

Do you mind to guild how to ping from the esp-sta to the notebook-sta?

### Code ( compiled with lwip v2 )
```cpp
#include <ESP8266WiFi.h>
//#include "lwip/lwip_napt.h"
//#include "lwip/app/dhcpserver.h"
//#include "lwip/ip_route.h"
#include "user_interface.h"
#include "ping.h"

#define  SSID         "-----------------"
#define  PASSWORD     "-----------------"

#define  AP_SSID      "-----------------"
#define  AP_PASSWORD  "-----------------"

// #define NAPT 1000
// #define NAPT_PORT 10

void esp_ping(IPAddress ip);

void setup() {
  Serial.begin(115200); Serial.println();
  WiFi.begin(SSID, PASSWORD);
  WiFi.softAP(AP_SSID, AP_PASSWORD);

//  ip_napt_init(NAPT, NAPT_PORT);
//  ip_napt_enable_no(SOFTAP_IF, 1);

  while(!WiFi.isConnected()) { delay(400); Serial.print("."); }
  Serial.println();
  Serial.print("STA IP : "); Serial.println(WiFi.localIP());
  Serial.print("AP  IP : "); Serial.println(WiFi.softAPIP());
  Serial.print("DNS IP : "); Serial.println(WiFi.dnsIP(0));

//  dhcps_set_DNS(WiFi.dnsIP(0));

  esp_ping(IPAddress(192,168,43,6));  // ping to IP of the Notebook-STA.
}

void loop() {
  // put your main code here, to run repeatedly:

}

void esp_ping(IPAddress ip){
  static struct ping_option  ping_opt;

  ping_opt.count        = 4;
  ping_opt.coarse_time  = 1;
  ping_opt.ip           = ip.v4();
  
  ping_regist_recv(&ping_opt, [](void* arg, void *pdata){
    struct ping_option* ping_opt = (struct ping_option*) arg;
    struct ping_resp *  ping_res = (struct ping_resp*) pdata;
      
    if(ping_res->ping_err == -1){
      Serial.printf("Reply from %s : offline\n", 
          IPAddress(ping_opt->ip).toString().c_str() );
    }else{
      Serial.printf("Reply from %s : bytes = %d time = %d ms\n", 
          IPAddress(ping_opt->ip).toString().c_str(), 
          ping_res->bytes, ping_res->resp_time);
    }
  });
  
  ping_start(&ping_opt);
}
```

### Result
[image]

TridentTD, 2020-03-02:
Another test with  `lwip-nat_arduino's liblwip_src.a`
for pinging from ESP-STA (192.168.43.254) to the Home Router (192.168.43.1),
the result is it can't pinging.